ConnectBox/connectbox-pi #27: Admin interface uses http

edwinsteele, 2016-12-11:
Admin credentials pass over cleartext, which is bad. We'll probably have to self-sign a certificate, which means that admin connections will get certificate warnings (perhaps there's another way?)

khoran, 2018-02-15:
I was just browsing through the issue list and thought I could add something to this discussion. You could use digest authentication, which only sends a hash of the password across the network. 
For apache, see here: http://httpd.apache.org/docs/2.2/mod/mod_auth_digest.xml
It's certainly not as good as a certificate, but better than plain text and won't scare the user as a self signed cert would.

joncojonathan, 2019-12-07:
Encouraging an end user to progress through a certificate prompt is a bad idea in my experience, as it breeds the viewpoint of "it was OK last time".

In the absence of a trusted certificate, a better option would be to SSH port forward the HTTP port.  This would allow the administrator to exchange their credentials without negatively impacting the user.  It should also be possible without any particular changes (I believe I read SSH was already available in the image).

Just a thought.

kirkdwilson, 2024-12-24:
We have a solution to the certificate. However, this requires internet access.  As of this moment we have chosen to hold on this as many users do not have internet access.

closed 2024-12-24 by kirkdwilson