ConnectBox/connectbox-pi #40: Allow setting of wifi tx power

edwinsteele, 2016-12-27:
The admin interface currently exposes a method to set the tx power but it’s not currently hooked up to do anything. It’s likely that we can do something with `iw dev <devname> set txpower <auto|fixed|limit> [<tx power in mBm>]`, but I don’t know how to make that stick across a reboot, short of putting it in an ifup type script.

I think it’s worth considering the use-cases for adjusting tx power - under what circumstances will the default be inappropriate, and can we simplify the interface so that it has low power (short range) and max power?

edwinsteele, 2016-12-27:
ok, so right after writing the ticket I did some testing and couldn’t improve the signal, as reported by iStumbler, by setting the txpower. I tried values from 10-1000000 using fixed i.e. `iw dev wlan0 set txpower fixed 1000000` and limit with a few more zeroes and the signal was no stronger than I got with `iw dev wlan0 set txpower auto`. I could definitely reduce the signal by using low values (<= 10), so I think that there’s only value in doing this work if there’s a clear use-case for wanting a weak wifi signal. If we don’t have that use-case I think we should remove the txpower UI control.

What do you think, @kldavis4 ?

kldavis4, 2016-12-28:
I think the only use case for lowering the signal power is if you wanted to evade detection beyond some range from the device. I say let's create an issue for it and remove from the UI for now until someone requests it.

edwinsteele, 2017-01-07:
We need a way to persist this setting, and until then we should grey out the TX Power menu entry.

PR #169 (2017-09-04): Remove txpower menu entry from admin interface
As our wifi interfaces run at max-power, as defined by the
regulatory domain, and there's no clear use-case for reducing
power, there's no need to expose setting txpower.

OK'ed by @geodirk

Fixes #40
commit d1f59264 (2017-09-04): Remove txpower menu entry from admin interface (#169)

As our wifi interfaces run at max-power, as defined by the
regulatory domain, and there's no clear use-case for reducing
power, there's no need to expose setting txpower.

OK'ed by @geodirk

Fixes #40
closed 2017-09-04 by edwinsteele