ConnectBox/connectbox-pi #41: Fix Android and Windows captive portal behaviour

edwinsteele, 2016-12-30:
We don’t have any captive portal logic for Android or Windows (phone/desktop) so I’m expecting that we’re still trapped inside the captive portal browsers (I don’t have devices to test, though)

The following resources should be sufficient to show what needs to be implemented:
- https://github.com/httpwg/wiki/wiki/Captive-Portals
- https://serverfault.com/questions/679393/captive-portal-popups-the-definitive-guide#
-  https://msdn.microsoft.com/en-us/library/windows/hardware/dn408681.aspx

kldavis4, 2017-01-18:
Tested on android and it thinks it is not connected and keeps asking to have the user login. It looks like for Android it is going to be somewhat involved, at least to cover a wide range of OS versions/variants. http://android.stackexchange.com/questions/123129/how-does-wifi-in-android-detect-if-the-device-has-to-sign-in-or-not

edwinsteele, 2017-01-28:
Windows looks for `http://www.msftncsi.com/ncsi.txt` . The file has special contents, but an approach like Apple’s `success.html` should be sufficient.
[Microsoft doc](https://technet.microsoft.com/en-us/library/cc766017(v=ws.10).aspx) also mentions DNS, but only in the context of Vista. Given the MSDN reference is about Win 8, and doesn’t mention the DNS method, it’s probably safe to assume that the nsci.txt method is the only one necessary for current clients.

edwinsteele, 2017-02-01:
Fixed by #69

closed 2017-02-01 by edwinsteele