ConnectBox/connectbox-pi #44: Match icons to font awesome directly

edwinsteele, 2016-12-30:
Currently we map key directory names to specific icons in the client interface. We will be providing instructions to connectbox administrators about how to layout files and directories on their USB stick place in order to take advantage of the iconography, so instead of constraining them to a small predefined mapping, let’s direct them to the [FontAwesome Icon List](http://fontawesome.io/icons/) and tell them that they can name their directories based on the FA icon names. e.g. a directory called paint-brush maps to http://fontawesome.io/icon/paint-brush (and the logic puts the element in the `fa-paint-brush` class).

@furnox / @kldavis4 , is one of you able to help with this?

Thanks for the suggestion, @geodirk

furnox, 2016-12-31:
I think we should be cautious about largely removing any trace of semantic data structure. Not only won't "paint-brush" mean anything to the next person to curate this data (without cross-referencing it with the FontAwesome list), but these most likely won't be native English speakers and thus it will easy for them to miss the connotation that could be derived from "paint-brush" (for, say, "Art").

Indeed, the icon we thought would best connote "Water" is actually the icon for "tint". That would be difficult for even an English speaker to connect.

edwinsteele, 2016-12-31:
I think we're in a difficult place and I'm grappling with a good approach tbh. A few things are guiding my thinking:
* people want to provide their own content and have structure in mind for presenting their content
* (for the reason above) the connectbox won't ship with content - we'll leave it to the connectbox owner/admin to manage it
* we'll have a large number of connectbox owners that have poor or non-existent English

The icon-only interface is great for users when the icons convey meaning, and it's terrible when it doesn't (another reason for a fallback textual folder names). We can't assume users will understand our semantic data structure (e.g. where textual items are placed under a folder called "written") so there's no downside to allowing them to pick a different icon that they think is suitable. Yeah, it's an extra step to go and find an icon, which makes me think that some predefined classification-icon mappings are helpful, but it's going to give flexibility which I think is really important.

My theory and thinking aside, however, we'll get awesome feedback once we put this in front of users, which is ultimately what we're driving for :-)

kldavis4, 2017-01-01:
I am a little late to this conversation but I'll throw in my 2 cents.. 

I wasn't aware that flexibility regarding folder names / icons was a requirement. My gut feeling on this is to start as simple as possible (with some fixed icons/folder names), unless it is seen by the group as a necessary thing. My recollection from the skype call was that internationalization on the admin side isn't something that is really needed, so it seems like that would extend to the folder structure on the USB drive. I may be remembering wrong, though. If it necessary to support something flexible like this, I think we need to write up the requirements and have some kind of basic design before starting on it, as it seems like it has the potential to quickly become complex.

kldavis4, 2017-01-18:
Implemented in this PR: https://github.com/ConnectBox/connectbox-pi/pull/65

closed 2017-01-18 by kldavis4