ConnectBox/connectbox-pi #45: Top level content and unmatched directories are not visible

edwinsteele, 2016-12-30:
It’s reasonable to expect connectbox owners to place files in the root directory of their USB stick, and to name their directories as something that doesn’t currently match our mapping (or that doesn’t correspond to a font-awesome icon once https://github.com/edwinsteele/biblebox-pi/issues/44 has been implemented).

- We should display these root directory files below the icons for the top-level directories.
- We should also display these unmatched folders (perhaps witha generic folder icon, or even the name directory given this is something that the user will have created and will be in an appropriate language - other suggestions?)

@furnox / @kldavis4 , are you interested in taking a shot at this? Better ideas on how to achieve this?

Thanks for the suggestion, @geodirk.

furnox, 2016-12-31:
Will the current Connect Box setup display arabic, korean, or hindi folder and file names? If it won't then you can't count on files being in an appropriate, meaningful language. And given that the genesis of the Bible Box was a method to convey universal meaning, the idea of a generic folder with no meaning doesn't make sense either.

I don't think we need to take the usual approach in this industry that users will try anything--these are people trying to convey information to the broadest slice of humanity possible. A few guidelines to follow on how to structure data isn't out of line. Not that the list of categories is exhaustive--it's the result of a brain-storming session at 2 a.m. and could very well be improved.

edwinsteele, 2016-12-31:
Thanks for your input. Users will be surprised when files and directories on their USB stick don't show in the interface, and given they'll be largely non-technical, I think flexibility in this area is important. I also think folder structure is one of those areas where it's hard to know what people will want to do so we should be cautious in enforcing hard rules. A few guidelines to help people along are appropriate, as you point out, and I also think sensible/unsurprising fallbacks are important for the inevitable times when people don't follow instructions or want to structure their folders differently.

I do prefer showing folder names when it doesn't match a known category, rather than a generic folder. We don't need to limit ourselves to icons, just prefer them.

Given the interface displays LtR UTF-8 files and directories nicely*, do you think folder names are inappropriate when it doesn't fit in an existing category? If not, what would you suggest?

* I found an issue with right-to-left issue languages when writing this, which we'll need to fix anyway

furnox, 2016-12-31:
Well, I don't think they'll be surprised when informed that anything in a non-standard category will not be displayed. I think non-technical folks would be less-likely to need flexibility in this area. I'm not usually surprised when my CD player fails to read upside-down discs. I note that the player failed to read my disc and then flip it over (fix my mistake).

I think that once you relax or remove the requirement for a rigid data structure, then why would anyone bother with it? What I don't know is if it's even important. If we're counting on content curators to name folders and files in a UTF-8 localized language, then maybe even the idea that we need language-less pictograms to convey universal meaning might be misplaced.

edwinsteele, 2017-01-01:
Yeah, it’s become clearer to me during this thread that we need to develop a bit more clarity around the pictograms use-case when we have local content curators, the extent to which we want to enforce a predefined mapping and semantic data classification. Obviously I think there’s value in being more flexible, and I thought I got that message from @GeoDirk too (feel free to clarify if I got that wrong, @GeoDirk )

I’ll chat with Elliott and @GeoDirk about it anyway, so let’s pause this until it’s clearer. Thanks heaps for the input @furnox .

K7TRY, 2017-01-03:
@GeoDirk has the most experience with non-technical end users who use languages other than English. I look forward to hearing what he has to say. Alex has experience as an ESL (English as a Second Language) user, but he is highly technical.

GeoDirk, 2017-01-04:
I think we need to expect to see two types of people in regard to content.  The first person will be the one that builds the initial structure for the USB stick.  This person we can expect to have a higher aptitude to figure things out and could expect him to be able to properly do a lookup on the Font Awesome icons to match the name of the icon with the directory.   He would be able to figure out that he would have to name the directory "headphones" if he wanted a directory that showed an icon for audio.  Of course, we would need to provide instructions for this step, but this won't be a problem even if the instructions are only in English.  These types of people have figured out how to read the instructions for flashing the original LibraryBox project and modifying the web content.

The next guy down the line will be the national worker who gets one of these boxes and decides to put in some additional content that he likes.  I would argue that this is the more important content as the national knows what files his people group is looking for.  Will he know that he should put a MP3 file into the directory called "headphones"?  Probably not right away but it doesn't seem that far fetched that he couldn't figure that out with a modest amount of poking around and looking at file types.  The bigger challenge comes when he has a new category and he decides to create a new folder called "книга" (which is Russian for 'book').  He puts his content in there and under the current build, it won't show up at all and he won't have a clue as to why or how to fix it.  This is unacceptable...either we need to put in a generic folder icon when the directory name doesn't match an FA icon or just put the word 'книга' in that spot.

As the point of this device is to share content, we need to ensure that people are able to get to the files regardless of how people end up slapping them on the USB stick; root folder or to some non-matching directory structure.

kldavis4, 2017-01-18:
Fixed in this PR https://github.com/ConnectBox/connectbox-pi/pull/65

closed 2017-01-18 by kldavis4