ConnectBox/connectbox-pi #60: Provide statistics on downloads

edwinsteele, 2017-01-13:
This is for administrators to see what files are being downloaded.

It might be something that we display in the admin interface (though perhaps it needs to be locked down more aggressively if that's the case).

It might also be something that's uploaded to a central site when an internet connection is detected. If this were the case, we could use a unique identifier present on the hardware (say a serial number, or the MAC of the wireliess interface) to identify the connectbox. We may want to disable upload stats by default, and we may want to show the link to the stat summary in the connectbox admin interface. We may want to hash the identifier if it helps with privacy.

edwinsteele, 2017-08-07:
@kldavis4 I think we can close this (given we’re not about to tackle uploads to a central site). Do you agree?

kldavis4, 2017-08-07:
So this is kind of half way implemented as a part of the popular downloads feature. There is a report in the admin ui that basically returns all of the historical download data and that is updated hourly along with the popular files. What is missing is the automatic upload functionality. So maybe make a new issue that is more about allowing configuration of an automatic upload if that is still something the consortium wants?

edwinsteele, 2018-01-05:
Closing this as we have good on-device reporting (per https://github.com/ConnectBox/connectbox-pi/pull/208). We can create a separate issue for centralised stats if /when that becomes a thing.

closed 2018-01-05 by edwinsteele