ConnectBox/connectbox-pi #93: Keyboard Support

GeoDirk, 2017-03-21:
By default, the Raspberry Pi's are coming with keyboards configured for the UK.  It would enhance our configuration if the keyboard layout was defaulted to the 'US' layout.

All that needs to be done is modify the one line in /etc/default/keyboard to look like this:

XKBLAYOUT="us"

edwinsteele, 2017-03-21:
@GeoDirk will probably have a go at this, when time permits. If it needs a fix before he can get to it, he'll let us know.

GeoDirk, 2017-03-24:
Submitted push request...

GeoDirk, 2017-03-24:
Code needs a bit of work...  Let me know what proper syntax is:

TASK [connectbox-pi : Set keyboard to the US layout] ***************************
fatal: [192.168.88.21]: FAILED! => {"changed": false, "failed": true, "msg": "unsupported parameter for module: path"}

PLAY RECAP *********************************************************************
192.168.88.21              : ok=50   changed=38   unreachable=0    failed=1

edwinsteele, 2017-06-21:
We decided not to proceed with this, as we’re not focussing on Raspbian.

closed 2017-06-21 by edwinsteele