ConnectBox/connectbox-pi #111: Migrate admin api from php to python

kldavis4, 2017-05-09:


portablejim, 2018-03-24:
Some work I have done on it https://github.com/portablejim/connectbox-pi/tree/python_admin

edwinsteele, 2018-03-24:
Hi @portablejim , thanks for your work on this, it looks great. We’ll need to break the commit up into a few PRs. Let’s start with the ansible bump (please PR that at your convenience) and over the next week I’ll spend some time to bring the other users of the connectbox virtualenv across to python3 (@kldavis4 this will be chat and captive portal). I’ve wanted to move these across to py3 for a while, and now I have an excuse :-) ). At that point we can get going on integrating your work and bid farewell to our php implementation. I’m travelling over this next week so my access may be spotty but we’ll get there.

PR #234 (2018-04-28): Complete removal of PHP
Also:
* Clarify test hosts during CI
* Fix noise during some admin actions due to gunicorn hosting both captive portal and admin API
* Add static site API test to improve coverage based on python API migration

Resolves #111 

FYI @portablejim @kldavis4
commit 6592dda5 (2018-04-28): Complete removal of PHP (#234)

* Remove remaining PHP config and tasks 
* Clarify test host during CI
* Fix noise during some admin actions due to gunicorn hosting both captive portal and admin API
* Add static site API test to improve coverage based on python API migration

Resolves #111
closed 2018-04-28 by edwinsteele