ConnectBox/connectbox-pi #118: Filter standard operating system folders on USB from icon interface

edwinsteele, 2017-06-13:
Originally mentioned by @algebur . A windows-formatted USB key has a folder called `System Volume Information` which shows in the interface. We should filter this out, as it’ll never have any end-user managed files.

Similarly for `.DS_Store` directories that might be present on USB keys used on a Mac.

GeoDirk, 2017-06-27:
Couple of standard windows ones:

-rwxr-xr-x 1 root root    16 May 26  2016 **AUTORUN.INF***
drwxr-xr-x 2 root root 65536 May 13  2016 **System Volume Information**/

Not sure why the autorun one ends in a *

edwinsteele, 2017-07-11:
Bumping from 25-unit pilot milestone after discussion with Dirk (Kelly’s rebuilding this area at the moment, and we can make it a “known issue” for the pilot)

edwinsteele, 2017-08-11:
Fixed by #152

closed 2017-08-11 by edwinsteele