ConnectBox/connectbox-pi #123: Decision: Which OS + Kernel combination for pilot

edwinsteele, 2017-06-21:
Mainline and Legacy kernel
Debian and Ubuntu OS

matheweis, 2017-06-21:
mainline has better drivers for the RALink/Mediatek chips.

An outstanding question was whether mainline supports USB on the Neo expansion headers.

Not aware of anything at this time that favors Ubuntu vs Debian from a technical standpoint.

edwinsteele, 2017-06-27:
I’m also not aware of any technical reason to prefer Ubuntu or Debian, but I do notice that Armbian are giving precedence to Ubuntu over Debian on their download page, so in the absence of anything else I suggest we go with Ubuntu.

I would be interested in comparing the base memory footprint between the two distros, in case one or other is starting up more daemons.

edwinsteele, 2017-06-27:
In subjective observations, mainline runs cooler (temperature as reported at login) and faster (time to login over ssh). If #124 was sorted, it’d be a no-brainer to go with mainline. I suspect #124 will frustrate a few people as they run initial tests.

edwinsteele, 2017-06-27:
Armbian don’t do mainline builds of the NEO for Debian, so if we want mainline we need to go with Ubuntu.

edwinsteele, 2017-06-28:
Once we have pilot hardware, we can do side-by-side testing for mainline vs legacy and debian vs ubuntu to ascertain whether there is any difference in performance/battery life.

In the meantime, let’s go with Ubuntu mainline because that seems like a sensible and forward-looking choice.

closed 2017-06-28 by edwinsteele