ConnectBox/connectbox-pi #124: Mainline kernel gives different eth0 MAC addresses on each boot

edwinsteele, 2017-06-21:
While this is certainly a pain while developing and testing, this might not be a problem for deployment if we don't expect people to access the device via ethernet interface.

matheweis, 2017-06-21:
On the flip side we may want to randomize the MAC address of the WiFi interface on each boot to improve privacy.

Perhaps resolving one issue may lead to a solution for the other?

edwinsteele, 2017-06-27:
I don’t see any benefit in randomising the MAC address of the Wifi Interface. We have a consistent SSID, so it’s clear that the device is serving the same purpose, even if the MAC address changes.

edwinsteele, 2017-08-31:
Fixed by https://github.com/armbian/build/commit/90b771cdf28b43cf6a857eeb10b204c38c7de82a which we’ll include in our image.

closed 2017-08-31 by edwinsteele