ConnectBox/connectbox-pi #149: Internal NEO USB ports stop working when kernel is updated

edwinsteele, 2017-07-15:
a `/boot/dtb` directory is created on each kernel update, however we create an overlay in that directory on our initial run in order to activate the internal USB ports. It’s hard to create that overlay on each kernel update, so we have a problem.

Thoughts:
* blacklist the kernel from being updated
* get the overlay added upstream. Even though the sun8i setup is shared with lots of boards, it may be accepted because the overlays aren’t activated without a line in armbianEnv.txt.

matheweis, 2017-07-15:
Solved w/easy fix to #150 imho, but will give it some thought.

edwinsteele, 2017-07-15:
Given the last step of our playbook is to update packages, I don't think #150 will solve this.

I deliberately put the overlay creation at the start of the playbook, so at worst we need to have one reboot that takes care of all the rebooty things like interface naming and overlay creation. This issue happens when the base image doesn't contain the latest kernel package, because the last step of our playbook updates it.

matheweis, 2017-07-15:
Can't we just run it again though?

edwinsteele, 2017-07-15:
Sure, but reboots increase the time and complexity of a deployment run so I think we should work hard to avoid them.

edwinsteele, 2017-08-12:
The overlays for usbhost0, 2 and 3 were added in https://github.com/armbian/build/commit/0630d4dcd9f1f7e995d9f7bca7d6a952ba5ce92c#diff-7a324c1a1d17511d2a3728f2cb90d449 . There’s no indication or comment on why usbhost1 was omitted.

edwinsteele, 2017-11-07:
The next step is to confirm that our apt update services (https://github.com/ConnectBox/connectbox-pi/blob/master/ansible/group_vars/armbian#L4) are indeed stopped and disabled when a device is booted from one of our images. Ansible does disable them, but I suspect that a first-run type service is re-enabling them.

PR #230 (2018-04-08): Move to armbian mainline 4.14
Overlays for usbhost0-3 exist by default (but are still not
enabled) so we no longer need to pin related packages or setup
the usbhost1 ourselves.

This requires our NEO armbian builds to be based off https://github.com/ConnectBox/armbian-build/releases/tag/base-image-180408 or newer

Fixes #149 #224
closed 2018-04-08 by edwinsteele