ConnectBox/connectbox-pi #156: Unable to set Wifi channel and SSID through admin interface

edwinsteele, 2017-08-10:
When updating the channel, the admin interface says: "Error updating channel” - Unexpected error setting property: error”. Same applies for updating the SSID, but setting the hostname is unaffected.

Tests to set SSID are still passing, and the problem does not appear on vagrant (which isn’t backed by a real interface). May be related to #145

edwinsteele, 2017-08-10:
Commandline exit status:
```
root@pilot1:~# /usr/local/connectbox/bin/ConnectBoxManage.sh get channel
5
root@pilot1:~# echo $?
0
root@pilot1:~# /usr/local/connectbox/bin/ConnectBoxManage.sh set channel 3
SUCCESS
root@pilot1:~# echo $?
0
root@pilot1:~# /usr/local/connectbox/bin/ConnectBoxManage.sh get ssid
new ssid
root@pilot1:~# /usr/local/connectbox/bin/ConnectBoxManage.sh set ssid pilot1.connectbox
SUCCESS
root@pilot1:~# echo $?
0
```

GeoDirk, 2017-08-15:
I noticed that even though it reports and error, it actually makes the change regardless.  Do you think this has anything to do with that the actual SSID has been modified and since you are connecting over WiFi, we end up not being connected which throws the error?  This would explain why doing over ethernet works and WiFi doesn't.

kldavis4, 2017-08-15:
Yes, that is exactly the issue. There may be something we can do in the admin ui to mitigate this.

edwinsteele, 2017-08-27:
Fixed by #162

closed 2017-08-27 by edwinsteele