ConnectBox/connectbox-pi #165: Admin interface allows invalid channel to be set (disables the AP)

edwinsteele, 2017-08-26:
Not all channels are available, depending on the regulatory domain. The admin interface allows channels to be set that prevent the AP from starting (even though the interface is up from the OS perspective).

kldavis4, 2017-08-26:
Interesting.. What's the best way to know what channels are valid?

edwinsteele, 2017-08-27:
There’s a regulatory db that specifies the valid channels for each regulatory domain, but there also seems to be some chipsets that simply don’t support all the channels. Still working out how that fits together, and what we can support. Perhaps on the back-end, we can rollback the change if hostapd doesn’t start?

Will get back to you on how we can determine what’s valid.

edwinsteele, 2017-08-30:
So the channel rules for the regulatory domain don't appear to tell the whole story, as there also appear to be constraints placed in the EEPROM of the wifi device. For example, with the reg domain set to AU, I should be able to setup an AP on channel 11, but the device prevents it.

I suggest we avoid client-side validation and make sure that we rollback the change to hostapd.conf if hostapd doesn't start. While it doesn't necessarily help the user in determining what's valid, we'll never brick the device.

(I'm avoiding client-side validation because a) it's hard to work out what is actually valid in the 2.4Ghz space, and b) when we start using 5Ghz channels the numbers are higher - and the interface shouldn't prevent someone from trying)

Sound ok?
--
Edwin Steele e: edwin@wordspeak.org w: https://wordspeak.org

kldavis4, 2017-08-31:
Yep, sounds like a good plan

edwinsteele, 2017-11-07:
I’ve not made progress on this, so a fallback is to constrain the channels to the intersection of those available in all reg-domains but turning the text field into a dropdown.

PR #195 (2017-11-26): Constrain channel to valid values in the UI
The valid channel list is conservative, but will be valid for all
regulatory domains, but ultimately we should retrieve the valid
channels from the API, so that we can include channels that are
valid for the current regulatory domain.

Fixes #165 (even though someone could still brick the box using the API directly, that’s not an important use-case at this stage)

I’m not sure whether my use of `$('#input_channel')[0].selectedIndex` is appropriate. I was surprised that looking up an element by ID returned a list, but I’m in unfamiliar territory so wa [... cut]
closed 2017-11-26 by edwinsteele