ConnectBox/connectbox-pi #190: Network LED's are on without connection on power up.

kirkdwilson, 2017-10-26:
When the unit powers up the network LED's are both on (yellow and green) consuming power.  While once a network jack is plugged in they indicate activity correctly, the power up state needs to be changed to off.  Also when the network cable is disconnected  they  revert to the on condition.  Again consuming power.

edwinsteele, 2017-10-26:
The long-term driver for the ethernet is the dwmac-sun8i which should land in kernel 4.14. Our kernel is 4.11, which has the discontinued sun8i-emac driver. A good course of action is to wait until the long-term driver lands in Armbian and go from there.

See for more details: https://linux-sunxi.org/Sun8i_emac

edwinsteele, 2017-11-30:
Fix should land in 4.15 per https://groups.google.com/forum/?_escaped_fragment_=topic/linux-sunxi/nucHmUshzfs#!topic/linux-sunxi/nucHmUshzfs