ConnectBox/connectbox-pi #204: Admin - need function to move files from USB to microSD

kirkdwilson, 2017-12-13:
For those who want to keep the unit simple and clean a file management tool that would allow for moving files from the USB flash to the microSD and/or deleting files on the disks is needed.

GeoDirk, 2018-03-29:
duplicate of #184

caroft, 2020-06-13:
> For those who want to keep the unit simple and clean a file management tool that would allow for moving files from the USB flash to the microSD and/or deleting files on the disks is needed.

that would be much easier, if possible... !
could you explain the documentation https://github.com/ConnectBox/connectbox-pi/blob/master/docs/deployment.md where it is said that : "If there are no reported errors, your RPi3 is now running as a ConnectBox. You can add your own content to it (e.g., videos, books, music, etc)."
How do you do that ? I tried a usb stick with some media content, but nothing works, I only see the content of /media/usb0 !
Thanks for your help !

kirkdwilson, 2022-09-12:
Done by the front panel.

closed 2022-09-12 by kirkdwilson