ConnectBox/connectbox-pi #205: Web Window title

kirkdwilson, 2017-12-13:
The basic web interface title in the browser will always show ConnectBox regardless of what the hostname is.  The Hostname should be used instead.

edwinsteele, 2017-12-26:
To be clear, when you say “web interface title” I’m assuming you mean the title that’s displayed in the browser e.g. _Web Window title - Issue #205 - ConnectBox/connectbox-pi_ for this page.

It’s unusual for the window title to be a hostname. Normally it’s something descriptive given the hostname is visible in the URL bar. Why should the hostname be the title?

kirkdwilson, 2017-12-26:
Generally I would agree.  However, since the host-name is the only way to control the 'personalization' of the graphical interface without web programming, it is generally used to 're-brand' the unit on the user interface.  If it remains ConnectBox/connectbox-pi then it becomes confusing to some users.

edwinsteele, 2017-12-27:
I think we’d do better with an admin option to change the page title rather than embedding the hostname.

kirkdwilson, 2022-09-12:
Done in latest releases using the brand.txt file

closed 2022-09-12 by kirkdwilson