ConnectBox/connectbox-pi #211: Create link to admin in icon-only interface

edwinsteele, 2017-12-29:
The admin area is important and needs to be discoverable. It isn’t as important as the chat function, so shouldn’t be given the same visibility, so let’s place a link (suggest a fa-cog icon) in a footer that’s a coloured div (no need to make it floating).

While it’s a little odd to be creating a footer for a single thing, we can probably use the footer area for other things once we see it in action.

kldavis4, 2017-12-31:
I've added the footer (and made a few other ui tweaks).  I haven't published the changes yet. When you get a chance, pull down the latest commit and run locally to see how it looks (`yarn start`). (Note the admin link won't actually take you to the admin dashboard). Once it looks good, I'll publish a new version of the ui.

edwinsteele, 2018-01-02:
Looks good, thanks. I have a slight preference for the icon being in the bottom right (it’s the least intrusive) but don’t feel strongly about it (leave it on the left if you’d like)

kldavis4, 2018-01-03:
Ok, thanks for the review. I've moved the icon to the bottom right and published the updated ui

edwinsteele, 2018-01-05:
Fixed by https://github.com/ConnectBox/connectbox-react-icon-client/commit/028210003c5d8395e31b64345a392730103aef17

closed 2018-01-05 by edwinsteele