ConnectBox/connectbox-pi #229: Reset to "factory condition"

rdai, 2018-04-01:
I got a connectbox last week. I think I have already lost the password, sorry! Is there a way to "factory reset" the unit so I can start over? Thanks

edwinsteele, 2018-04-01:
Hi @rdai . Did you change the password? If not, it’ll be admin/connectbox per the [admin docs](https://github.com/ConnectBox/connectbox-pi/blob/master/docs/administration.md). If you have changed it, you’ll need to download an image and apply it to your microSD card. Are you using the storage on the microSD for your media, or are you using external USB storage? (so we can work out what image you will need)

rdai, 2018-04-01:
I think I did change the password. I think it sat there for awhile when I tried to change password and I had to repeat it few times so when it took I didn’t pay attention. Sorry!

 

I am using the external USB that comes with it. Is it possible to have access to both just in case?

 

Thanks you.

Ray

edwinsteele, 2018-04-04:
No problem. There are a few steps (a few more than we’d like to be honest, but we’re working on the update mechanism).

1. Download the latest release image from GitHub: https://github.com/ConnectBox/connectbox-pi/releases/download/v20180122/nanopi-neo_release_180122.img.7z
2. Uncompress it.
3. Download Etcher (http://etcher.io/) so you can burn the image to the microSD. We recommend this program even if you have another method for burning, as it actively verifies that the image has been successfully burnt.
4. Take the microSD card out of the ConnectBox (you’ll need to open the case to do this)
5. Burn the image to the microSD card using Etcher.
6. Pop it back in the ConnectBox and start it.

You won’t be able to use the internal storage on the microSD for data with this process, but as you’re using external USB storage, that shouldn’t be a problem. There is a process that allows you to use it, but it’s for advanced users: https://connectbox.technology/wp/utilizing-your-existing-microsd-boot-card-for-your-media-files/

Please let me know how you go.

edwinsteele, 2018-04-16:
Hi @rdai . How did you go?

edwinsteele, 2018-05-02:
Timing out this request from inactivity. @rdai feel free to reopen if you still need input.

closed 2018-05-02 by edwinsteele

rdai, 2018-05-03:
Edwin,

Sorry for the late reply. I was able to reflash this successfully just now, thank you!

edwinsteele, 2018-05-03:
Glad to hear it, Ray. You’re welcome!