ConnectBox/connectbox-pi #233: Folder Icon Rendering Issues

GeoDirk, 2018-04-25:
We have several methods for rendering a file folder on the GUI interface.  Several of them currently have some issues:

Method 1: name your folder the same as a Font Awesome name (e.g, 'heart').  Yes, it does work but only on the root level directory.  **This needs to be fixed so that it is on all sub directories as well.**

Method 2: give your folder name whatever you want (people) and then create an empty file that points it to a Font Awesome icon name.  So if I make a file called "anyfilename" and then make an empty file that is named: _icon_anyfilename_heart, I would expect that the 'anyfilename' folder now looks like the heart icon.  **It doesn't and just returns back an empty/blank icon spot - so this needs to be fixed as well.**

Method 3: create any folder name (I used the numbers one, two, three, four) and then create images that match those folder names in the format _icon_one.png, _icon_two.png, etc,.  This works just fine and consistently worked in subfolders down to the three levels I tested to.  **No issues with this method.**

Method 4: don't put in any associated _icon... file/image so you get the standard default Font Awesome icon always works just fine.  **No issues with this method.**

Test set of examples attached (unzip to root directory of USB stick):

[folder_test_set.zip](https://github.com/ConnectBox/connectbox-pi/files/1945722/folder_test_set.zip)

kldavis4, 2018-04-25:
I've published a new version of the UI that fixes the issues with method 1+2. For method 2, it was expecting the 'fa-' prefix on the icon class name. I modified it to prefix it with 'fa-' if that is not already present to handle both cases.

edwinsteele, 2018-05-05:
Fixed by https://github.com/ConnectBox/connectbox-react-icon-client/commit/c48ec969eca9b3ab1cdee1ad7849c4d5c61e15a3

closed 2018-05-05 by edwinsteele