ConnectBox/connectbox-pi #236: Consider Perkeep as a backend

glowingwire, 2018-05-07:
https://github.com/perkeep/perkeep

This project does a lot of things that we would need too.  It may be good only for back end stuff.

edwinsteele, 2018-05-08:
spam

closed 2018-05-08 by edwinsteele

glowingwire, 2018-05-09:
may have been unwanted and premature, but certainly I meant well by posting that.

perkeep does transfering of files by USB, and can use many other file transfer methods... which appears to be a complicated part of connectbox...  and connectbox also does the serving of the files for people who have limited connectivity.    

I'm just saying that these people have done a lot of work on ssl encryption and other things that are hard...  and we can just focus on the simple UI and deployment of the system.  The files could be shared in ways other than with perkeep.. but perkeep sort of keeps it all together.

Back in the day I considered modifying bittorrent to use instead of rsync.   This perkeep project seems to take care of that.  If you start transfering files over a slow dial-up modem,  those can start being distributed, then when someone comes with a USB drive and plugs it into part of the network, it can be propagated to the rest of the network.

I'm kinda sad that this was taken as spam because I spent a lot of time contemplating the project, and see a very robust system that is free for us to use.

edwinsteele, 2018-05-09:
Thanks for the explanation. I closed this issue because it referenced a tool that would require significant rework to integrate. There was also very little detail in the ticket, combined with a person that I didn't know who doesn't have a commit history. I appreciate your thought. Perkeep is now in my head in case it intersects with something that we do in the future.

glowingwire, 2018-05-11:
great,
I can see how it was unsolicited and out of place,

Am excited by the project for transfering files and making sure information
can be shared.