ConnectBox/connectbox-pi #245: Chrome DNS Issues with Android 8

kldavis4, 2018-06-04:
I'm seeing a strange issue with the latest release(s) on my Android phone where it is not able to resolve the `connectbox` host. The error in the browser indicates a DNS resolution failure. I confirmed that this is not occurring on my older Android devices. Do you have any recommendations on how to troubleshoot?

edwinsteele, 2018-06-05:
Is this with a built image on a device that doesn't have any internet connection? (remembering that you've had problems with DNS when running ansible playbooks). Some android devices are fussy with DNS (see https://github.com/ConnectBox/connectbox-pi/blob/master/ansible/roles/dns-dhcp/templates/etc_dnsmasq.conf.j2#L20) so wanting to rule out any on-device changes.

kldavis4, 2018-06-05:
Yes, built image without internet, running that admin ui update branch.

kldavis4, 2018-06-07:
More data on this.. When I get the sign in prompt and it brings up the captive portal browser, I select Use Network As Is, which clears it. The WIFI indicator on the top of the screen looks normal and I am able to load the UI and login. Then shortly after I start getting the DNS errors. I notice also that the WIFI has a little X mark on it, which indicates that the device doesn't think it has a fully baked internet connection. I do not get a new captive portal sign in prompt. So it looks like latest Android is doing some additional connectivity checks aside from those that trigger the captive portal sign in, and then apparently disabling internet in some fashion when those fail. I tried using Android Firefox as well and it appears to be affected in the same way - so not specific to Android Chrome.

kldavis4, 2018-06-07:
I think I figured out what the issue is.. Looks like it is detecting the bad network over wifi and falling back to the cellular data connection without bothering to ask :/ Then it is no longer using the CBX dns, resulting in the DNS failure.

I think ultimately we need to figure out what Android 8 is doing to check internet connectivity that earlier versions did not - but at least I know how to workaround it for the moment.

edwinsteele, 2018-07-24:
Fixed with #250

closed 2018-07-24 by edwinsteele