ConnectBox/connectbox-pi #252: JPG / PNGs don't show up in Android 4.x

GeoDirk, 2018-07-27:
On Android 4.x phones with the Icon-Only interface, images do not show up - this includes regular images as well those associated with folders.  Seems to me that there is a loading issue rather than they cannot just display the images.

Root folder contents:
[image]

Android 7.0 Render:
[image]

Android 4.0 Render:
[image]

Also note that on the Android 4 phone, the "C" icon in the upper left just spins and spins around carrying the smaller "refresh" icon around with it.

GeoDirk, 2018-07-27:
BTW - tried on several other phones as well.  Discovered that I could trigger the same thing on a 5.x phone if I used the stock browser.  If I used Chrome on that device, it rendered just fine.

GeoDirk, 2018-07-30:
I had the original submitter retest their Android 4 device by installing and using the Chrome browser.  This is the reply back:

_I tested the connect box again with colleagues phones. When they used Chrome, they could see the icons and download the files! (They are getting very excited about using the box!)

Hope this helps._

larsgson, 2018-08-24:
I have now managed to setup a useful debugging environment. Here is the error message I get:
   _Console: Uncaught TypeError: Object / has no method 'startsWith'_

So I checked on the internet and here I have links with descriptions of similar problems:
  a possible workaround in code: https://stackoverflow.com/questions/646628/how-to-check-if-a-string-startswith-another-string
  or possibly here: https://stackoverflow.com/questions/33106114/cordova-javascript-error-has-no-method-startswith-android

I think the easiest solution would be a simple polyfill fix within npm: 
https://github.com/DasRed/js-string.startsWith-polyfill

Since I so far only set up a debugging environment, not a complete build environment, then I think it will be easier if somebody else just tries if the above solves the problem. If not I can still try to dig deeper within my debugging environment.

larsgson, 2018-09-05:
As far as I can tell from my testing, we should now be able to close this issue. I suppose this is part of the latest test release, so @GeoDirk have you been able to verify if it is now working for you by using this release?

edwinsteele, 2018-09-08:
The spinning "C" problem is gone but there are other rendering problems on my 4.2.2 device that prevent some files from appearing in the file listing for icon-only interface. Will post images in a followup.

edwinsteele, 2019-05-31:
Our minimum supported version is now Android 4.4, so we won't do anything more on the 4.2 rendering failures.

closed 2019-05-31 by edwinsteele