ConnectBox/connectbox-pi #323: Provide 2D or 3D cad for PCBA

montieroland, 2021-11-05:
Having the CAD files makes it easier to package the unit in a custom enclosure

kirkdwilson, 2021-11-29:
I'll check into this.

montieroland, 2022-03-08:
Morning,

Do you have a new release that I can install on my ConnectBox?

Montie

Montie Roland
***@***.***
Sanford, NC

A candle loses nothing by lighting another candle

Blog and Podcast - www.montie.com
Montie Gear - www.montiegear.com
Simply Us Farm - www.simplyusfarm.com

montieroland, 2022-03-08:
> Dear Montie;
>
> Yes I do. I will compile up a new release which we are no presently testing. AS soon as it goes live I’ll let you know.
>
> Kirk

Thanks

Sent from ProtonMail mobile

-------- Original Message --------

kirkdwilson, 2022-03-10:
Dear Montie;

 

Please look at this release:

https://github.com/ConnectBox/connectbox-pi/releases/tag/v20220309.2.0

 

This is hot off the press and includes the new interface and admin console

 

Kirk

montieroland, 2022-03-11:
Were you able to create a 3D cad file of the electrical assembly?  Or a drawing of the assembly?

kirkdwilson, 2022-03-12:
> Hi,
>
> Do I still need to follow the "Configuring a Connectbox for WordPress" directions to use WordPress?

Are you planning to put wordpress on it?

 

Kirk

kirkdwilson, 2022-03-12:
Is this for the Connectbox Pi or the Connectbox NEO?

 

Kirk

montieroland, 2022-03-12:
Yes

Sent from ProtonMail mobile

-------- Original Message --------

montieroland, 2022-03-12:
Whichever one you provided

Sent from ProtonMail mobile

-------- Original Message --------

closed 2022-09-12 by kirkdwilson