ConnectBox/connectbox-pi #336: LoRa capability

montieroland, 2022-03-11:
Have you considered add LoRa capability so the boxes could form or participate in an ad hoc network?  Maybe talk at a Meshtastic network (https://meshtastic.org/docs/getting-started)

kirkdwilson, 2023-05-25:
Yes.  WE have test hardware and are considering this verses 802.11ah.   LoRa is a very slow protocol so we are concerned about the value add.  the 802.11ah ;is a fast protocol.  It has more value add.

montieroland, 2023-05-27:
I can see that. I think LoRa comes in when you want to network ConnectBoxes, maybe an integration with MeshTastic for emergency info or for info across a large campus.

Montie

Montie Roland
***@***.***
Sanford, NC

A candle loses nothing by lighting another candle

Blog and Podcast - www.montie.com
Montie Gear - www.montiegear.com
Simply Us Farm - www.simplyusfarm.com

kirkdwilson, 2024-12-24:
We are considering 802.11ah as it has higher bandwidth.   In a mesh network this is critical.

montieroland, 2024-12-25:
If you could add Meshtastic compatibility that could open up a new market that is growing quickly