ConnectBox/connectbox-pi #338: Create a tourism friendly version

montieroland, 2022-11-27:
Here is a possible user journey for an extended chat feature (chat with host)

- visitor connects to connectbox
- visitor downloads documents relevant to the venue
- visitor has question so he clicks on the chat button in browser on phone
- chat request is routed through a gateway (internet or cellular) to host's cell phone (text message) and email
- host replies to chat request via text message and it is routed back to the visitor that is physically at the connectbox
- visitor receives reply via his browser on his phone
- chat continues as needed

This might work well for a campground / campsite

montieroland, 2023-05-27:
That sounds great.

Montie

Montie Roland
***@***.***
Sanford, NC

A candle loses nothing by lighting another candle

Blog and Podcast - www.montie.com
Montie Gear - www.montiegear.com
Simply Us Farm - www.simplyusfarm.com

montieroland, 2024-12-24:
Did you ever publish app for a tourism friendly version?

kirkdwilson, 2024-12-24:
We have not created an app to be used for tourism as the .ain goal of connectbox is not to need an app to access it.  You can connect to connectbox by scanning a barcode to simplify the connection and access.

montieroland, 2024-12-24:
Should be barcode direct the user to gowifi.org? Have you already generated a barcode?

Montie Roland
***@***.***
Sanford, NC

A candle loses nothing by lighting another candle

Blog and Podcast - www.montie.com
Montie Gear - www.montiegear.com
Simply Us Farm - www.simplyusfarm.com

montieroland, 2024-12-24:
The intent wasn't to request a phone app, instead streamline the login, but I like the idea of a qr code

Montie Roland
***@***.***
Sanford, NC

A candle loses nothing by lighting another candle

Blog and Podcast - www.montie.com
Montie Gear - www.montiegear.com
Simply Us Farm - www.simplyusfarm.com

kirkdwilson, 2024-12-24:
We ha e not added the extended chat functionality unless you are using the server integrated version.

closed 2024-12-24 by kirkdwilson