ConnectBox/connectbox-pi #339: Auto reset feature

montieroland, 2022-11-28:
If the unit loses the Wifi connection for y minutes, it auto restarts x times with an xx minute wait to try and reconnect.

Would this help if router reset for some reason?

kirkdwilson, 2023-05-25:
We have a monitor application in place.  It monitors the Wifi from the services and SSID as well as the dhcpcd.  It attempts to fix these issues in real time.  A timeout function may be useful for a forced reboot.

montieroland, 2023-05-27:
A forced reboot would be great.

Here is where I want to deploy the box:

www.troublesomegap.com
www.simplyusfarm.com

Thanks,
Montie

Montie Roland
***@***.***
Sanford, NC

A candle loses nothing by lighting another candle

Blog and Podcast - www.montie.com
Montie Gear - www.montiegear.com
Simply Us Farm - www.simplyusfarm.com

closed 2024-12-24 by kirkdwilson