ConnectBox/connectbox-pi #342: Still need easer acess

Ham549, 2023-08-11:
Library Box another open source media platform is set up to automatically bring you to the main page as soon as you connect to the network. It would be nice if ConnectBox could do the same. Also, it would be nice to be able to leave out the most recent downloads and comment function.

montieroland, 2023-08-11:
Agreed

Montie Roland
***@***.***
Sanford, NC

A candle loses nothing by lighting another candle

Blog and Podcast - www.montie.com
Montie Gear - www.montiegear.com
Simply Us Farm - www.simplyusfarm.com

-------- Original Message --------

closed 2024-12-24 by kirkdwilson

Ham549, 2025-01-24:
It still doesn't seem to work, It would also be nice if there was a desktop included So instead of having to SSH into everything I can just use it as a standalone computer and transfer the file through USB. I had previously had a bill that had a desktop included And it made it very easy to add and remove content. Also would still like to know how to remove the chat function because it's out being able to monitor the connect box all the time I'd hate for somebody to leave something offensive on it.