ConnectBox/connectbox-pi #343: "Indexing Data" message peristent forever at startup

montieroland, 2024-12-24:
I'm getting an "Indexing Data" message when I start up the unit running  the 20241011_armbian24.8.5.img.xz version.  Any suggestions?

montieroland, 2024-12-24:
I'm running a connectbox P and all three lights are lit (yellow, orange, red)

montieroland, 2024-12-24:
I'm able to connect the wifi of the box, but not able to connect to the gowifi.org site

closed 2024-12-24 by kirkdwilson

montieroland, 2024-12-25:
[image]

What else do you need?

kirkdwilson, 2024-12-25:
It would appear that the connectbox is not transmitting correctly on your PC.  Can you show me the network screen

Kirk

Screenshot.2024-12-24.193637.jpg (view on web)<https://github.com/user-attachments/assets/d1dcfa8d-792a-4bce-b2ba-27800645fc5c>

montieroland, 2024-12-25:
Where is the network screen?

Montie Roland
***@***.***
Sanford, NC

A candle loses nothing by lighting another candle

Blog and Podcast - www.montie.com
Montie Gear - www.montiegear.com
Simply Us Farm - www.simplyusfarm.com

-------- Original Message --------